gussmith23/lakeroad #344: Sheep-and-goats problem

gussmith23, 2023-08-17:
Claire proposed an example project:

>Also, I've been thinking a bit more about the use case for lakeroad that we chatted about that n our last conversation, to position it more as a design tool than as s synthesis tool, for cases where the designer wants to build something that is supposed to map to a (combination of) DSP slices.
>
>It would actually be interesting to use that flow, to map Verizon code onto an entire Xilinx L-slice or M-slice, or half or a quarter of slices.
>
>That could be extremely useful for designing things like ALU bitslices, where each bitslice is supposed to map to a small fixed fraction of the FPGA fabric, in a highly optimized way.
>
>I even have a possible example project: a while ago I have found a way of implementing the so-called sheep-and-goats operation, using an array of similar cells, and it would be interesting to see if that can be mapped to a slice, maybe even utilizing features such as the carry chain elements in a creative and non-obvious way.

gussmith23, 2023-08-17:
We need to bug Claire for the sheep-and-goats code. She has it, she just needs to dig it up (and she needs us to bug her about it).

PigBrainOverflow, 2025-05-18:
Wait! I guess she's wanting ArchHDL. :)