gussmith23/lakeroad #458: Importing bsg_defines.sv causes Lakeroad to fail

cknizek, 2024-08-19:
Reproducing issue:

Including the file [bsg_dff_chain.sv](https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_dff_chain.sv) from BaseJump STL in a Lakeroad integration test will cause it to fail.

To reproduce this issue:
> 1. Add the file `bsg_dff_chain.sv` to the local directory of an integration test
> 2. Insert the line `"include bsg_dff_chain.sv"` at the top of the file, after the section of Lakeroad `// RUN:` statements
> 3. In the top-level directory, run the command `docker build -t lakeroad .`
> 4. Once the Docker build is finished, run the command `docker run lakeroad`

Once I run the 4 steps above, I receive the following terminal output [1].

I only receive an error when including the line `include bsg_dff_chain.sv`. If I only insert the line `include bsg_defines.sv`. So, it seems like some files are okay to include and some are not.

I'm still trying to figure out what's going on, but it seems like this might be related to the [issue with module boundaries](https://github.com/uwsampl/churchroad/issues/91).

I think this because (from what I've gathered), inserting an `include` statement in Verilog/SystemVerilog will basically copy + paste the contents of the included file directly into the file where the `include` statement exists.

And so, if there is an existing issue with multiple modules within the same file, it might cause conflict between the modules. 

The reason [my PR](https://github.com/uwsampl/lakeroad/pull/454) was passing locally was likely because I was setting up the test incorrectly (i.e. using `lit -v bsg_mul_add_unsigned.sv` instead of re-building the Docker container and running it from there). 

[1] > `******************** TEST 'Lakeroad tests :: example_test_using_include.v' FAILED ********************
Script:
--
: 'RUN: at line 1';   outfile=$(mktemp)
: 'RUN: at line 2';   racket $LAKEROAD_DIR/bin/main.rkt   --solver bitwuzla   --verilog-module-filepath /root/lakeroad/integration_tests/lakeroad/example_test_using_include.v   --architecture xilinx-ultrascale-plus   --template dsp   --out-format verilog   --top-module-name combinational_multiplier   --verilog-module-out-signal p:16   --pipeline-depth 0   --module-name out   --input-signal a:16   --input-signal b:16   --timeout 90  > $outfile
: 'RUN: at line 16';   FileCheck /root/lakeroad/integration_tests/lakeroad/example_test_using_include.v < $outfile
: 'RUN: at line 17';   if [ -z ${LAKEROAD_PRIVATE_DIR+x} ]; then    echo "Warning: LAKEROAD_PRIVATE_DIR is not set. Skipping simulation.";    exit 0;  else    python3 $LAKEROAD_DIR/bin/simulate_with_verilator.py     --test_module_name out     --ground_truth_module_name combinational_multiplier     --max_num_tests=10000     --verilog_filepath $outfile     --verilog_filepath /root/lakeroad/integration_tests/lakeroad/example_test_using_include.v     --pipeline_depth 1     --output_signal p:16     --input_signal a:16     --input_signal b:16     --verilator_include_dir "$LAKEROAD_PRIVATE_DIR/DSP48E2/"     --verilator_extra_arg='-DXIL_XECLIB'     --verilator_extra_arg='-Wno-UNOPTFLAT'     --verilator_extra_arg='-Wno-LATCH'     --verilator_extra_arg='-Wno-WIDTH'     --verilator_extra_arg='-Wno-STMTDLY'     --verilator_extra_arg='-Wno-CASEX'     --verilator_extra_arg='-Wno-TIMESCALEMOD'     --verilator_extra_arg='-Wno-PINMISSING';  fi
--
Exit Code: 1`

gussmith23, 2024-08-19:
This is caused at least partly by not having a `--top-module` flag being passed to Verilator. Verilator is throwing a multiple top modules warning, causing an error. We can just statically pass `--top-module testbench`.

gussmith23, 2024-08-19:
@cknizek you can add `--top-module testbench` to verilator.mk.template?

cknizek, 2024-08-21:
> @cknizek you can add `--top-module testbench` to verilator.mk.template?

Done!